AIR-THU/DAIR-V2X #8: Error about Create Kitti-format data for sv3d-inf

shb9793, 2022-07-25:
When I ran the following command:
```
python tools/dataset_converter/dair2kitti.py --source-root ./data/DAIR-V2X/DAIR-V2X-I/single-infrastructure-side \
--target-root ./data/DAIR-V2X/DAIR-V2X-I/single-infrastructure-side \   --split-path ./data/split_datas/single-infrastructure-split-data.json \   --label-type lidar --sensor-view infrastructure
```

The error was encountered as follows:
```
================ Start to Convert ================
================ Start to Copy Raw Data ================
================ Start to Generate Label ================
Traceback (most recent call last):
  File "tools/dataset_converter/dair2kitti.py", line 80, in <module>
    json2kitti(json_root, kitti_label_root)
  File "/home/shb/open-mmlab/dair-v2x/tools/dataset_converter/gen_kitti/label_json2kitti.py", line 36, in json2kitti
    write_kitti_in_txt(my_json, path_txt)
  File "/home/shb/open-mmlab/dair-v2x/tools/dataset_converter/gen_kitti/label_json2kitti.py", line 22, in write_kitti_in_txt
    i15 = str(-item["rotation"])
TypeError: bad operand type for unary -: 'str'
```

shb9793, 2022-07-25:
```
File "/home/shb/open-mmlab/dair-v2x/tools/dataset_converter/gen_kitti/label_json2kitti.py", line 22, in write_kitti_in_txt
    i15 = str(-item["rotation"])
```

>>>>>>>>

```
 i15 = str(-eval(item["rotation"]))

`item["rotation"]` is a `str` type, not `float` type.

closed 2022-07-25 by shb9793

Drafaa, 2022-12-19:
some thing still wrong with 
` i15 = str(-eval(item["rotation"]))`
because `item["rotation"]` is a `float` type in my computer, to avoid this question. I just using 
`i15 = str(-float(item["rotation"]))`
to ensure the type of `item["rotation"]`.