AIR-THU/DAIR-V2X #12: Error occured when reimplementing pointpillar on vic3d dataset

CatOneTwo, 2022-08-02:
Hi, I  have followed the **configs/vic3d/late-fusion-pointcloud/pointpillars/README.md** to prepare the dataset, but when I try to train PointPillar model with trainval_config_i.py, the error occurred as
 `No such file or directory: '../../../../data/DAIR-V2X/cooperative-vehicle-infrastructure/infrastructure-side//kitti_infos_train.pkl`.

I have checked the converted dataset, which indeed doesn't contain `kitti_infos_train.pkl`. So how can I generate this file, thank you!

yuhb94, 2022-08-09:
You should convert the dataset format as MMDetection3D requiring. More details refer to https://mmdetection3d.readthedocs.io/en/latest/data_preparation.html.

closed 2022-08-12 by yuhb94

deyang2000, 2024-02-25:
> Hi, I have followed the **configs/vic3d/late-fusion-pointcloud/pointpillars/README.md** to prepare the dataset, but when I try to train PointPillar model with trainval_config_i.py, the error [...]
[image]

I had the same problem. Did you solve it? I saw the corresponding download link in the mmdet3d manual, but I don't think it's right.

deyang2000, 2024-03-04:
I had some possible solutions but didn't implement them because I switched to using the V2X-set data set for now. The general idea is to follow the documentation for custom datasets in mmdet3d.

Ericchen328, 2025-01-13:
请问有什么解决方案吗