AIR-THU/DAIR-V2X #18: Some infrastructure pcd file missing in the cooperative label_world

smallscientist1, 2022-08-18:
When using this dataset, I found that in ./cooperative-vehicle-infrastructure/cooperative/data_info.json file, the pcd file in "infrastructure_pointcloud_path" attribute  sometimes doesn't exist at all. Is this a problem of the dataset?

haibao-yu, 2022-11-25:
> When using this dataset, I found that in ./cooperative-vehicle-infrastructure/cooperative/data_info.json file, the pcd file in "infrastructure_pointcloud_path" attribute sometimes doesn't exist at all. Is this a problem of the dataset?

Can you provide more details? For example, which idx of the infrastructure_pointcloud_path.

haibao-yu, 2023-02-28:
This is normal. We associate vehicle-side and infrastructure-side frames according to a certain time difference constraint. There are some vehicle-side frames at the edge of the clip without corresponding infrastructure-side frames

closed 2023-02-28 by haibao-yu