AIR-THU/DAIR-V2X #20: KeyError: 'KittiMultiViewDataset is not in the dataset registry'

JiayinDeng, 2022-11-07:
When using Imvoxnet, the config file in `configs/sv3d-inf/imvoxelnet/trainval_config.py` uses `KittiMultiViewDataset` dataset type. However, the official version of mmdetection3d v0.17.1 doesn't have this type of dataset. In addition, I noticed that the Imvoxelnet in https://github.com/saic-vul/imvoxelnet contains `KittiMultiViewDataset` dataset, while the mmdetection3d v0.8.0 version is required.

haibao-yu, 2022-11-26:
> When using Imvoxnet, the config file in `configs/sv3d-inf/imvoxelnet/trainval_config.py` uses `KittiMultiViewDataset` dataset type. However, the official version of mmdetection3d v0.17.1 doesn't [...]

Thanks for your remind. This config is suitable for the order version of mmdetection3d. You can also directly use the imvoxelnet config in mmdetection3d 0.17.1, which now can only support one class. We are collaborating with the mmdetection3d to build the new config of multiple class for imvoxelnet.

haibao-yu, 2022-12-05:
> When using Imvoxnet, the config file in `configs/sv3d-inf/imvoxelnet/trainval_config.py` uses `KittiMultiViewDataset` dataset type. However, the official version of mmdetection3d v0.17.1 doesn't [...]

We have updated the ImvoxelNet config, checkpoint, and evaluation results with mmdetection3d==0.17.1. More details can refer to the commit https://github.com/AIR-THU/DAIR-V2X/commit/60332a6e8a32345941e1b4b9c19b71d5ab05468c.

closed 2022-12-05 by haibao-yu