AIR-THU/DAIR-V2X #30: some problem happen when convert point cloud data from infrastructure coordinate system to vehicle coordinate system.

linjie111, 2022-12-05:
the link is here [https://github.com/AIR-THU/DAIR-V2X/tree/main/configs/vic3d/early-fusion-pointcloud/pointpillars](url)

An error occurred when the third step was executed ， I ran all night, but the next day it was like this. I'm so sad，could you tell me what happen? Thank you very much ，the wrong screenshot is below.
[image]

koi-boy, 2022-12-11:
Did you fix this problem?

15838127929, 2022-12-16:
The script handles all the data, and since the test dataset we have not yet released, using the script directly will report an error. We suggest you filter the ‘data_info.json’ with https://github.com/AIR-THU/DAIR-V2X/blob/main/data/split_datas/cooperative-split-data.json.

15838127929, 2022-12-16:
> The script handles all the data, and since the test dataset we have not yet released, using the script directly will report an error. We suggest you filter the ‘data_info.json’ with https://github.com/AIR-THU/DAIR-V2X/blob/main/data/split_datas/cooperative-split-data.json.

The script handles all the data, and since the test dataset we have not yet released, using the script directly will report an error. We suggest you filter the ‘data_info.json’ with https://github.com/AIR-THU/DAIR-V2X/blob/main/data/split_datas/cooperative-split-data.json.

15838127929, 2022-12-16:
> the link is here [https://github.com/AIR-THU/DAIR-V2X/tree/main/configs/vic3d/early-fusion-pointcloud/pointpillars]
> 
> An error occurred when the third step was executed ， I ran all night, but the next day it was like this. I'm so sad，could you tell me what happen? Thank you very much ，the wrong screenshot is below. [image]
> 
> [image]

The script handles all the data, and since the test dataset we have not yet released, using the script directly will report an error. We suggest you filter the ‘data_info.json’ with https://github.com/AIR-THU/DAIR-V2X/blob/main/data/split_datas/cooperative-split-data.json.

15838127929, 2022-12-16:
> Did you fix this problem?

The script handles all the data, and since the test dataset we have not yet released, using the script directly will report an error. We suggest you filter the ‘data_info.json’ with https://github.com/AIR-THU/DAIR-V2X/blob/main/data/split_datas/cooperative-split-data.json.

weihongwei-zg, 2023-01-17:
How to filter?

weihongwei-zg, 2023-01-17:
@YingjuanTang

sogoojoy, 2023-01-29:
There is no '006315.pcd' file in infrastructure-side folders, we can just delete '006315.pcd''s imformation from cooperative-vehicle-infrastructure/cooperative/data_info.json