AIR-THU/DAIR-V2X #56: Table 5 in your paer, the vehicle type

wanghangege, 2023-04-23:
Hello, thank you for your outstanding work, but could you please answer my question? Has "Vehicle", "Pedestrian", and "Cyclist" been redefined? like behind, because I try the dair-v2x-i detection in openpcdet with pointpillar, but got different results.
[image]
[image]
my results of  the dair-v2x-i detection in openpcdet with pointpillar:
[image]

wanghangege, 2023-04-23:
And if the type of "Vehicle", "Pedestrian", and "Cyclist" have been redefined, could you please give me the definition you set in somewhere.

longlonglonghouhouhou, 2023-05-19:
请问下，您是按照官方说明的步骤将数据集转化为kitti格式的吗？我按照官方说明处理后，使用openpcdet  kitti_dataset.py 提取的gt_database里的点云文件都是空的

wanghangege, 2023-06-04:
> Hello, thank you for your outstanding work, but could you please answer my question? Has "Vehicle", "Pedestrian", and "Cyclist" been redefined? like behind, because I try the dair-v2x-i detection in openpcdet with pointpillar, but got different results.

是啊，应该是你的读取或者写出有问题吧，很多基于官方说明进行转化的仓库都有实现，并且成功，可以参考类似的工作