AIR-THU/DAIR-V2X #60: DAIR数据集转成KITTI数据集后车头位置有问题

Jarvisxize, 2023-05-26:
DAIR的路侧数据转成KITTI数据后通过可视化展示后，发现车头的位置顺时针偏了90度，即车头都变成了车的右侧，这个问题如何解决

Ignite616, 2023-07-29:
> 

我也遇到了相同的问题,请问您怎么解决的

yaobaishen, 2024-04-30:
in my case, the converted kitti format label is NOT on the same horizontal plane with the lidar pointcloud, which leads to the label unusable, anyone see this issue?
[image]

yaobaishen, 2024-04-30:
> in my case, the converted kitti format label is NOT on the same horizontal plane with the lidar pointcloud, which leads to the label unusable, anyone see this issue? ![Screenshot from 2024-04-30 15-05-59](https://private-user-images.githubusercontent.com/8359082/326703896-f71af6a7-9703-4ac3-a9dc-3ea0a06e2bed.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ0NzA4NTgsIm5iZiI6MTcxNDQ3MDU1OCwicGF0aCI6Ii84MzU5MDgyLzMyNjcwMzg5Ni1mNzFhZjZhNy05NzAzLTRhYzMtYTlkYy0zZWEwYTA2ZTJiZWQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQzMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MzBUMDk0OTE4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YTllYWM2NjAyYWQzYjg2OWUxOTFhMjYyYmZkOGFiMWE3YWViNDU5YTRkZmJhZDczMmQzZjdjMWExMGUxODUyMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.G6bUdKmvCQqOryVlVXeIZH9eIr62EL8HyAGL_r-qA4A)

please ignore this... it's a bug in the visualization tool, not the kitti converter.
I have filed an issue in the visualization tool retpo: https://github.com/zzzxxxttt/simple_kitti_visualization/issues/2