AIR-THU/DAIR-V2X #64: Question about rotation matrix from virtual lidar to camera

huanghuanghuanghaung, 2023-06-16:
Hello, contributors of DAIR dataset, thanks for your great work. 

There is a issue that confuses me when I use the dataset. I'll be very appreciate if you can answer my question. 

When I looked into the rotation matrix from virtual lidar to camera, I found the rotation matrix is not a standard rotation matrix since norm of every column and raw of the rotation matrix doesn't equal to 1 and the rotation matrix doesn't meet the orthogonality condition. But when I transfer ground truth from virtual lidar coordinate system to camera coordinate system, it works well.
So I wonder where does the rotation matrix come from and how do you get it.

You can check the rotation matrix from virtual lidar to camera of frame 000009 to find out what is mentioned above.

WeiSQ-zju, 2023-12-14:
I have the same question.

haibao-yu, 2024-01-14:
> Hello, contributors of DAIR dataset, thanks for your great work.
>
> There is a issue that confuses me when I use the dataset. I'll be very appreciate if you can answer my question.
>
> When I looked into [...]

Thanks for your question. We conducted a secondary calibration to generate a new transformation matrix, aiming to rectify errors present in the original calibration. The oversight of omitting constraints led to a non-standard rotation matrix. This issue can be resolved through the generation of a standard rotation matrix approximation.

closed 2024-01-14 by haibao-yu