AIR-THU/DAIR-V2X #74: Dataset request -- V2X-Seq -- Outside Mainland China

DerejeCShenkut, 2023-10-05:
Hi,

I have repeatedly sent an email to dair@air.tsinghua.edu.cn requesting a download link for research outside China. 
Can you please respond?

Thanks,

haibao-yu, 2023-11-03:
Thank you for your interest in our dataset. Please note that, at present, the dataset is officially available for download and use exclusively within Mainland China.

To help researchers become acquainted with our data, we have provided a Google Drive link for direct download of an example dataset. 
 - DAIR-V2X-C-Example: [google_drive_link](https://drive.google.com/file/d/1y8bGwI63TEBkDEh2JU_gdV7uidthSnoe/view?usp=drive_link)
 - V2X-Seq-SPD-Example: [google_drive_link](https://drive.google.com/file/d/1gjOmGEBMcipvDzu2zOrO9ex_OscUZMYY/view?usp=drive_link)
 - V2X-Seq-TFD-Example: [google_drive_link](https://drive.google.com/file/d/1gjOmGEBMcipvDzu2zOrO9ex_OscUZMYY/view?usp=drive_link)

 If you wish to utilize our complete dataset, we encourage collaboration with researchers based in Mainland China.

closed 2023-11-03 by haibao-yu

tugbakara, 2024-04-14:
Hi @haibao-yu , I also send the e-mail but noone turned back, how can I do to reach the full dataset?