AIR-THU/DAIR-V2X #75: Training      Implementation Framework. We directly use MMDetection3D (v0.17.1) to train the infrastructure 3D detector and vehicle 3D detector.      Infrastructure detector training details. Before training the detectors, we should follow MMDetection3D to convert the "./data/DAIR-V2X/cooperative-vehicle-infrastructure/infrastructure-side" into specific training format. Then we train the PointPillars with configure file trainval_config_i.py      Vehicle detector training details. Before training the detectors, we should follow MMDetection3D to convert the "./data/DAIR-V2X/cooperative-vehicle-infrastructure/vehicle-side" into specific training format. Then we train the PointPillars with configure file trainval_config_v.py

Ting234, 2023-10-19:


Ting234, 2023-10-19:
[Training](https://github.com/AIR-THU/DAIR-V2X/blob/main/configs/vic3d/late-fusion-pointcloud/pointpillars/README.md#training)

    Implementation Framework. We directly use MMDetection3D (v0.17.1) to train the infrastructure 3D detector and vehicle 3D detector.

    Infrastructure detector training details. Before training the detectors, we should follow MMDetection3D to convert the "./data/DAIR-V2X/cooperative-vehicle-infrastructure/infrastructure-side" into specific training format. Then we train the PointPillars with configure file [trainval_config_i.py](https://github.com/AIR-THU/DAIR-V2X/blob/main/configs/vic3d/late-fusion-pointcloud/pointpillars/trainval_config_i.py)

    Vehicle detector training details. Before training the detectors, we should follow MMDetection3D to convert the "./data/DAIR-V2X/cooperative-vehicle-infrastructure/vehicle-side" into specific training format. Then we train the PointPillars with configure file [trainval_config_v.py](https://github.com/AIR-THU/DAIR-V2X/blob/main/configs/vic3d/late-fusion-pointcloud/pointpillars/trainval_config_v.py)
怎么训练呢？该怎么操作呢！求

haibao-yu, 2024-01-13:
> Then we train the PointPillars with configure file [trainval_config_i.py]怎么用呢

You can convert the DAIR-V2X dataset into Kitti format as https://github.com/AIR-THU/DAIR-V2X/tree/main/configs/sv3d-inf/pointpillars, and then train and eval the model as mmdetection3d https://mmdetection3d.readthedocs.io/en/latest/user_guides/train_test.html.

closed 2024-01-13 by haibao-yu