AIR-THU/DAIR-V2X #82: Questions about the SPD installation process

deyang2000, 2023-12-27:
[image]
[image]
I find that the versions of MMDET3D 0.17.1 and AB3DMOT Dependencies are completely different when I try to install them. How do I solve this problem? Thank you very much.

yangzvv, 2024-01-22:
Hi there! To address the compatibility issue between MMDET3D 0.17.1 and AB3DMOT, you may want to try using the following versions of dependencies, which are known to be compatible:

- mmdet3d==0.17.1
- scikit-learn==1.0.2
- filterpy==1.4.5
- numba==0.48.0
- matplotlib==3.5.2
- pillow==9.4.0
- opencv-python==4.7.0.72
- pyyaml==6.0
- llvmlite==0.31.0
- wheel==0.38.4

Please make sure to install these versions and check if the issue persists. If you encounter any specific errors or issues during installation, feel free to provide more details, and the community can assist you further.

Hope this helps! Let me know if you have any more questions.

deyang2000, 2024-01-31:
Thanks for your reply. I will try relevant suggestions and continue to ask questions if there are any questions. Thank you very much.