AIR-THU/DAIR-V2X #84: 车路协同3D检测数据集 infrastructure-side 中image 的6143--6687缺失

happyman000009, 2024-01-12:
车路协同3D检测数据集（即DAIR-V2X中的DAIR-V2X-C数据集） infrastructure-side 中image 的6143--6687缺失，我是在（[DAIR-V2X车路协同数据集 (baai.ac.cn)](https://thudair.baai.ac.cn/coop-dtest)网址下载的全部数据集，不是DAIR-V2X-C-Example，是DAIR-V2X-C）。是原本数据集的设定就是这样吗？如果不是，我应该去哪里下载这部分数据呢
[image]
？谢谢。

sidiangongyuan, 2024-02-20:
请问你可以提供数据集吗？ 我发现官网无法下载了已经

happyman000009, 2024-02-22:
抱歉，才看到。你好，我试了下，官网还是可以下载的。你在中国大陆地区应该是没什么问题。建议使用edge浏览器注册下载。

sidiangongyuan, 2024-02-22:
十分感谢，已经下载了。  另外还请问一下， 官方提供的cooperative-split-data这个json文件里面包括了 batch_split\vehicle_split\infra_split\cooperative_split ， 如果我只作协同感知，是不是只需要cooperative_split 部分？ 
我将这部分划分成train val test，但是发现test有的编号并没有在 cooperative 的 data_info中出现。 不知道有没有类似的情况？

sidiangongyuan, 2024-02-22:
> 
> 十分感谢，已经下载了。 另外还请问一下， 官方提供的cooperative-split-data这个json文件里面包括了 batch_split\vehicle_split\infra_split\cooperative_split ， 如果我只作协同感知，是不是只需要cooperative_split 部分？ 我将这部分划分成train val test，但是发现test有的编号并没有在 cooperative 的 data_info中出现。 不知道有没有类似的情况？

 已解决。

hy-cn, 2024-03-01:
请问问题解决了吗，车端的部分数据标号也有缺失，路侧的数据总量是12424，车端的则是15285，数据集车端与路侧的数据不是一一对应的吗？

image1432, 2024-10-18:
> 车路协同3D检测数据集（即DAIR-V2X中的DAIR-V2X-C数据集） infrastructure-side 中image 的6143--6687缺失，我是在（[DAIR-V2X车路协同数据集 (baai.ac.cn)](https://thudair.baai.ac.cn/coop-dtest)网址下载的全部数据集，不是DAIR-V2X-C-Example，是DAIR-V2X-C）。是原本数据集的设定就是这样吗？如果不是，我应该去哪里下载这部分数据呢？谢谢。

您好，我在做数据集转化的时候也遇到了同样的问题（6143--6687缺失），请问这个问题你已经解决了吗？

happyman000009, 2024-10-26:
> > 车路协同3D检测数据集（即DAIR-V2X中的DAIR-V2X-C数据集） infrastructure-side 中image 的6143--6687缺失，我是在（[DAIR-V2X车路协同数据集 (baai.ac.cn)](https://thudair.baai.ac.cn/coop-dtest)网址下载的全部数据集，不是DAIR-V2X-C-Example，是DAIR-V2X-C）。是原本数据集的设定就是这样吗？如果不是，我应该去哪里下载这部分数据呢？谢谢。
> 
> 您好，我在做数据集转化的时候也遇到了同样的问题（6143--6687缺失），请问这个问题你已经解决了吗？

目前还没有，可以尝试联系下数据集的作者的邮箱，我记着貌似数据集下载网站上有，好像我之前也发了邮件，但是没有回复

wissage, 2025-01-06:
您好，我目前尝试了好几个浏览器都无法登上数据集官网，请问您方便分享一下数据集吗，谢谢

2429993608, 2025-03-26:
> > 
> > 
> > 十分感谢，已经下载了。 另外还请问一下， 官方提供的cooperative-split-data这个json文件里面包括了 batch_split\vehicle_split\infra_split\cooperative_split ， 如果我只作协同感知，是不是只需要cooperative_split 部分？ 我将这部分划分成train val test，但是发现test有的编号并没有在 cooperative 的 data_info中出现。 不知道有没有类似的情况？
> 
> 已解决。

> > 
> > 
> > 十分感谢，已经下载了。 另外还请问一下， 官方提供的cooperative-split-data这个json文件里面包括了 batch_split\vehicle_split\infra_split\cooperative_split ， 如果我只作协同感知，是不是只需要cooperative_split 部分？ 我将这部分划分成train val test，但是发现test有的编号并没有在 cooperative 的 data_info中出现。 不知道有没有类似的情况？
> 
> 已解决。

请问，您是如何解决“test有的编号并没有在 cooperative 的 data_info中出现”这个问题的？是重新划分了数据集吗

ysy200187, 2025-06-24:
> 目前还没有，可以尝试联系下数据集的作者的邮箱，我记着貌似数据集下载网站上有，好像我之前也发了邮件，但是没有回复

请问你解决了吗，我遇到了同样的问题