AIR-THU/DAIR-V2X #107: Dataset download from outside China

hp2002xx, 2025-02-10:
@haibao-yu 
Currently, you are providing DAIR-V2X datasets at the link below:
https://drive.google.com/drive/folders/1gnrw5llXAIxuB9sEKKCm6xTaJ5HQAw2e?usp=sharing

Since this link is accessible from anywhere, the datasets can be downloaded from outside China, which is not allowed until recently.
Then, I have some questions.

1. Is it allowed to download datasets from outside China, currently?
2. If it is allowed, can I write academic papers showing experiments using the datasets?

I would be very grateful if I could use the datasets.

AnRabbit, 2025-02-11:
Hello, could you please tell me where I can find the download link for the new dataset? I would be very grateful!

AnRabbit, 2025-02-11:
> [@haibao-yu](https://bgithub.xyz/haibao-yu) Currently, you are providing DAIR-V2X datasets at a new download link.目前，您在新的下载链接上提供了DAIR-V2X数据集。 Since this link is accessible from anywhere, the datasets can be downloaded from outside China, which is not allowed until recently.由于这个链接可以从任何地方访问，数据集可以从中国以外的地方下载，直到最近才被允许。 Then, I have some questions.然后，我有一些问题。
> 
> 1. Is it allowed to download datasets from outside China, currently?目前是否允许从中国境外下载数据集？
> 2. If it is allowed, can I write academic papers showing experiments using the datasets?如果允许，我可以写学术论文展示使用数据集的实验吗？
> 
> I would be very grateful if I could use the datasets.如果我能使用这些数据集，我会非常感激。

Hello, could you please tell me where I can find the new download link for the  dataset? I would be very grateful!

AnRabbit, 2025-02-11:
@hp2002xx 
Hello, could you please tell me where I can find the new download link for the dataset? Could you provide the download link, please?I would be very grateful!

Uchan1996, 2025-02-11:
You can find the download link in the README.
Specifically, the name of download link is "Public-V2X-Datasets", and the link is in the "Dataset Download" section.
However, I'm not sure if it is allowed to download the datasets from outside China.
If you live in China, I think there is no problem.

hp2002xx, 2025-02-11:
@AnRabbit 
You can find the download link in the section of "Dataset Download" in the README.
But, I'm not sure if it is allowed to download the datasets from outside China.
If you live in China, I think there is no problem.

haibao-yu, 2025-02-21:
Please refer to the latest download link at [README](https://github.com/AIR-THU/DAIR-V2X).

closed 2025-02-21 by haibao-yu

hp2002xx, 2025-02-21:
@haibao-yu 
I understand this dataset can be used for academic activities outside of China since any restrictions are not stated in the download link and README.
I will download this dataset and use it on my academic activities.
Please let me know if there are any issues.
Thank you for your publicly providing DAIR-V2X datasets.