AIR-THU/DAIR-V2X #109: Inconsistent # images & point clouds

ChengChen2020, 2025-02-12:
Hi,

In the paper, the number for DAIR-V2X-C images and point clouds are both 39k.

[image]

However, from the downloaded dataset at [link](https://drive.google.com/drive/folders/1gnrw5llXAIxuB9sEKKCm6xTaJ5HQAw2e), I got the following statistics

- \# Vehicle image: 15285
- \# Infra image: 8577
- \# Vehicle point cloud: 15285
- \# Infra point cloud: 12424

The number does not match and the infrastructure side data sizes are inconsistent.

Could you please check and confirm?

Thanks!

haibao-yu, 2025-02-21:
Thanks for your feedback. We have updated the DAIR-V2X-C, and we do not release the test part. Please check it again.

ChengChen2020, 2025-02-23:
Thanks for your reply. However, the statistics I got remain the same. Which files did you update?

zhang-huizhi, 2025-04-07:
I have the same problem, the google drive version seems to be missing a lot of camera files for infra side.

haibao-yu, 2025-04-09:
Thanks for your valuable feedback, we have updated the DAIR-V2X-C infrastructure images with 12424. Note that we have not released the test part. Please refer to [V2X-Public-Dataset ReadMe](https://docs.google.com/document/d/1PlQyltoo4lLGRGPLx-TmMpzbDaihI-kswzwGWY3YvB8).

closed 2025-04-09 by haibao-yu