AIR-THU/DAIR-V2X #115: dair2kitti swaps length and width?

jonathan-kiener, 2025-03-25:
It is my understanding of the KITTI format that the labels are ordered as follows:

```
<object_type> <truncation> <occlusion> <alpha> <left> <top> <right> <bottom> <height> <width> <length> <x> <y> <z> <rotation_y> <score?>
```

The dair2kitti file seems to very clearly swap width and length. This is a clear mistake, right? The correct version even exists in a comment above.

[image]

lacie-life, 2025-04-21:
Have you successfully converted data to Kitti format? @jonathan-kiener . After the DAIR roadside data was converted into Kitti data and visualized, it was found that the front of the car was 90 degrees clockwise, which means that the front of the car became the right side of the car. How to solve this problem?

akml2013, 2025-04-25:
> Have you successfully converted data to Kitti format? [@jonathan-kiener](https://github.com/jonathan-kiener) . After the DAIR roadside data was converted into Kitti data and visualized, it was found [...]
 According to my observation. When you converted data to Kitti, the yaw is still in the Lidar coordinate system(It seems to have taken a negative value).